microsoft/mu_plus #102: UefiTestingPkg-AuditTests-PagingAudit needs more enhancements to make sense of the data

spbrogan, 2022-11-12:
This issue is tracking numerous minor enhancements that can be made to the Audit tool to help it be more useful. 

1. "Special Memory Usage" shows heap guard page guards.  
  1. It doesn't show NullPtr (page 0) guard page. 
  2. It doesn't show stack guard pages.
  3. Are there others.  On QemuQ35 i see 4 guard pages.  I expect that the two cases above cover at least 2 or 3 of them.

spbrogan, 2022-11-12:
Documentation about using the DxeDriver and UefiApp needs updating.  

1. The options around collecting the data at time of executing the app vs collecting the data the dxe driver saved at some other point in boot needs to be clarified.  
2. The details about how files were written to FS0 in the dxe driver case doesn't make sense.  
3. Dxe Version of app should save data into the dat files to indicate at what point in the boot process was this environment audit collected.

spbrogan, 2022-11-12:
GCD Memory Type is listed as None for all.  Is something missing?

TaylorBeebe, 2022-12-19:
> GCD Memory Type is listed as None for all. Is something missing?

Fixed here: https://github.com/microsoft/mu_plus/pull/128

TaylorBeebe, 2023-04-05:
The following PRs address the feature requests:

https://github.com/microsoft/mu_plus/pull/131
https://github.com/microsoft/mu_plus/pull/132
https://github.com/microsoft/mu_basecore/pull/216
https://github.com/microsoft/mu_plus/pull/196

closed 2023-04-05 by TaylorBeebe