microsoft/mu_plus #225: [Feature]: UefiTestingPkg binary releases

mhaeuser, 2023-04-20:
### Feature Overview

UefiTestingPkg provides various conformance tests for the UEFI environment. These are at least mostly generic and may be used on any UEFI implementation, including such not based on Project Mu. However, its source code has dependencies on Project Mu packages and thus cannot easily be built without a separate development environment.

For this reason, binary releases would greatly improve the usability of UefiTestingPkg outside of Project Mu platforms. Unfortunately, UefiTestingPkg.dsc is explicitly not designed to produce usable binaries as of now:

https://github.com/microsoft/mu_plus/blob/79dd13a1088fcc4dab256951bb6b7ad7b10f1c21/UefiTestingPkg/UefiTestingPkg.dsc#L18-L20

Many of the dummy library classes are effectively generic. These include:
- MemoryAllocationLib
- DebugPrintErrorLevelLib
- PcdLib

One more library class could be, but currently sometimes may not be [1]:
- HobLib

The one library class that cannot really be generic as of now is DebugLib. While this should usually not pose a problem, the default instance for the UnitTestResultReportLib library class actually is UnitTestResultReportLibDebugLib:

https://github.com/microsoft/mu_plus/blob/79dd13a1088fcc4dab256951bb6b7ad7b10f1c21/UefiTestingPkg/UefiTestingPkg.dsc#L58

This means that there is no trivial way to build universal binary releases.

[1] In case a non-Null DebugLib is required, ArmVirtPkg has an ArmVirtDxeHobLib that does not depend on DebugLib. The reason is the risk for circular dependencies, such as the ArmVirt serial code that depends on HobLib: 

https://github.com/microsoft/mu_silicon_arm_tiano/blob/922860bc4cb762e229b441de8a4ed2b127070168/ArmVirtPkg/Library/FdtPL011SerialPortLib/FdtPL011SerialPortLib.inf#L25

There is nothing ARM-specific about this approach and it might as well be DxeHobLibNoDebug variant (which actually could share its core code with the regular DxeHobLib).

### Solution Overview

As I don't have a good overview of everyone's workloads and use cases, I can't propose a solution myself. Potential approaches I can think of are:

- Using UnitTestResultReportLibConOut for binary builds. This would trivially be universal and require only a small patch to UefiTestingPkg.dsc. However, I'm not sure parsing outputs to ConOut is common practice and well-supported. I'm not aware something like this exists, but it may require something to re-route ConOut outputs to a variety of targets, such as a serial port (which we currently use for our CI, as it can easily be routed to stdout via QEMU).

- Rather than hooking ConOut, a test result protocol can be defined by the platform ahead of launching the test application to route the test result to any arbitrary target. This would require a new instance of UnitTestResultReportLib and a driver to expose said protocol.

- The protocol could be more "generic" in the sense that it may be used for other kinds of output. For OpenCorePkg, we have a centralized logging protocol that is also utilized by DebugLib:

https://github.com/acidanthera/OpenCorePkg/blob/2e93c954023df44d7d68c4d896c32a5d07439b3c/Include/Acidanthera/Library/OcLogAggregatorLib.h

This leaves the OpenCore application in charge of the logging targets and the OpenCore drivers consuming its logging protocol. We use this approach to decide where to route logging output based on a user configuration file, which may include multiple targets at once. Naturally, a new DebugLib instance would need to be introduced. However, in this scenario, UnitTestResultReportLibDebugLib could be re-used as-is.

### Urgency

Low

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

spbrogan, 2023-04-21:
@mhaeuser

Thanks for the report and glad to see someone is looking into using some of these tests as they have helped us identify and maintain fixes for a lot of issues with industry reference code.  

The problem you call out is a general one that we are finding is growing as we author more tests.  Many (not all) UEFI shell-based tests really need to be compiled in an environment that matches the platform they are going to run on.   An example of our effort is https://github.com/microsoft/mu_plus/pull/210 which is trying to untangle most of our memory audit test from project mu changes so that the binary can execute on a platform that may not have all the changes.  This effort also tried to make the app compile in an environment using EDK2 rather than Project Mu (although this has led to significantly more complexity in the app source/build files).  This was again for the same reason you identified and trying to make these test applications more easily used by the diverse and inconsistent ecosystem.  

Regarding DebugLib I am suggesting to my team that if we have some apps that can be compiled for compatibility that we should use the debugport instance of debuglib which is in Edk2 and Project Mu.  
* https://github.com/tianocore/edk2/tree/master/MdePkg/Library/UefiDebugLibDebugPortProtocol
* https://github.com/microsoft/mu_basecore/tree/release/202208/MdePkg/Library/UefiDebugLibDebugPortProtocol

We have used this approach with OpRom vendors in the past and it has worked well but I don't think it is "industry standard".  In Project Mu we also have code that produces DebugPort by our DxeCore and leverages whatever the DebugLib solution used for DxeCore. 

Regarding Package DSC files.  For Project Mu package DSC files are not intended for anything other than CI testing and for CI testing it is best to leverage as many NULL libraries as possible to avoid unnecessary dependencies and build time.   So, I am opposed to changing out the library mappings in the package dsc file.  

That said, I would like to find ways to make it easier for platforms to find and use UEFI shell-based tests that are implemented in core packages and other repositories.  In Edk2, we have discussed the usage of "DSC inc" files.  I don't like these as I find additional complexity in DSC files generally create more problems than they solve.  This also would not solve your problem about dependencies and code base environments.  For select UEFI shell modules in UefiTestingPkg which can be generalized for execution on a "UEFI system" we could add an additional DSC file and/or release pipeline.  Would that work?  Given the comments written above do you have other ideas?  

thanks

mhaeuser, 2023-04-21:
@spbrogan Thanks for the swift response!

> Regarding DebugLib I am suggesting to my team that if we have some apps that can be compiled for compatibility that we should use the debugport instance of debuglib which is in Edk2 and Project Mu.

I completely forgot about that protocol, that'd basically be the third option I listed. Whether this is "clean" is kind of debatable, as this mixes things potentially used in an production environment with debug semantics, but we do not actually use this protocol for anything, so, pragmatically, this would resolve the issue for us.

> Regarding Package DSC files. For Project Mu package DSC files are not intended for anything other than CI testing and for CI testing it is best to leverage as many NULL libraries as possible to [...]

I understand the idea, but sorry, this sounds very odd. Ideally, there would be some CI mode that allows to do this for critical library instances automatically (I hope most packages get more than only build-testing?). This just makes the package a pain to use to save probably less than 1 second of build time (at least with Clang :) ).

> For select UEFI shell modules in UefiTestingPkg which can be generalized for execution on a "UEFI system" we could add an additional DSC file and/or release pipeline. Would that work?

Yes, that would be great (including the pipeline), thanks!

closed 2023-07-24 by github-actions[bot]