microsoft/mu_plus #237: [Feature]: Augment AdvancedLogger on ARM devices to allow retrieval of in-memory UEFI boot logs

singweeMsftGitHub, 2023-05-11:
### Feature Overview

AdvancedLogger allows retrieval of UEFI boot logs on x86 devices in two fashions:

1. UefiLogs directory on EFI system partition or USB storage.
2. In-memory logs.

On ARM devices, only the first method (UefiLogs directory method) is available.

This request is for feature parity between x86 and ARM devices.

### Solution Overview

Would like ARM platforms to have the ability to retrieve in-memory UEFI logs, similar to x86 devices.

The varstore methodology of storing logs may need to be revisited.

### Urgency

Medium

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

makubacki, 2023-05-19:
Assigned to @TaylorBeebe

TaylorBeebe, 2024-04-11:
This feature has been handled without the need for mu_plus changes. Closing this issue.

closed 2024-04-11 by TaylorBeebe