microsoft/mu_plus #309: [Bug]: AdvancedLoggerHdwPortLib should not check PCD for header versions greater than 2

spbrogan, 2023-09-06:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

wasted execution time on a hot path comparing the debug level against the PCD at both levels.   
https://github.com/microsoft/mu_plus/blob/a3d53ef6296bc0b1309cf2654017eb394a726dd4/AdvLoggerPkg/Library/AdvancedLoggerHdwPortLib/AdvancedLoggerHdwPortLib.c#L64

Why does calling code just mask the bits rather than avoid the function call?
https://github.com/microsoft/mu_plus/blob/a3d53ef6296bc0b1309cf2654017eb394a726dd4/AdvLoggerPkg/Library/AdvancedLoggerLib/AdvancedLoggerCommon.c#L148

### Expected Behavior

Expected to be as efficient as possible and require minimal code in lower layers. 

### Steps To Reproduce

Look at code

### Build Environment

```markdown
NA
```

### Version Information

```text
a3d53ef6296bc0b1309cf2654017eb394a726dd4

all versions.
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

May have impact on downstream/out-of-tree projects and their implementations.

os-d, 2023-09-07:
There is a case here where we could not have LoggerInfo, in the SEC instance:

https://github.com/microsoft/mu_plus/blob/a3d53ef6296bc0b1309cf2654017eb394a726dd4/AdvLoggerPkg/Library/AdvancedLoggerLib/AdvancedLoggerCommon.c#L141-L153.

So, either we need the HdwPortLib to check the PCD and the upper lib not to, or the SEC instance needs to be reworked or have a separate code path. The smaller delta here is having the HdwPortLib check the PCD and being unconditionally called, so I'll put up a PR for that and we can discuss if a different approach is better.

os-d, 2023-09-11:
Looking again, if we unconditionally log in SEC, which seems reasonable, we can drop this double checking.

PR #311 (2023-10-10): Only call HdwPortWrite if DebugLevel Met
## Description

The DebugLevel is checked twice in the hot path on serial path writes and the HdwPortWrite call is made even if the upper layer knows that the message being logged does not meet the DebugLevel criteria.

Closes #309.

In order to maintain backwards compatibility, if the LoggerInfo block is found to have a version less than the hardware logging level version, the PCD is checked to decide whether to call HdwPortWrite or not.

In SEC, because we may not have the LoggerInfo structure, we check the PCD to see if the message should be logged a [... cut]
commit 4c5d1d1c (2023-10-10): Only call HdwPortWrite if DebugLevel Met (#311)

## Description

The DebugLevel is checked twice in the hot path on serial path writes
and the HdwPortWrite call is made even if the upper layer knows that the
message being logged does not meet the DebugLevel criteria.

Closes #309.

In order to maintain backwards compatibility, if the LoggerInfo block is
found to have a version less than the hardware logging level version,
the PCD is checked to decide whether to call HdwPortWrite or not.

In SEC, because we may not have the LoggerInfo structure, we check the
PCD to see if the message should be  [... cut]
closed 2023-10-10 by kuqin12