microsoft/mu_plus #531: [Bug]: <MemoryOutsideEfiMemoryMapIsInaccessible() Test Is Broken>

TaylorBeebe, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

[PR 528](https://github.com/microsoft/mu_plus/commit/69e9464117eff84e88b9646c0e9c2e95a807fe60) updated the MemoryOutsideEfiMemoryMapIsInaccessible() test to check the return status of ValidateRegionAttributes(). **ValidateRegionAttributes() returns a boolean, not a status.** So, interpreting the FALSE return value is being mistaken for an EFI_SUCCESS return value.

I know, I know... usually we return a status... SOMETIMES I COLORED OUTSIDE THE LINES 😅

### Expected Behavior

[The AllowUnmappedRegions argument of ValidateRegionAttributes()](https://github.com/microsoft/mu_plus/blob/release/202311/UefiTestingPkg/AuditTests/PagingAudit/UEFI/Dxe/App/DxePagingAuditTestApp.c#L508) dictates if EFI_NO_MAPPING is acceptable for the region. So, the MemoryOutsideEfiMemoryMapIsInaccessible() test was already checking for EFI_NO_MAPPING but now the test will only pass if the required conditions **are not** met.

### Steps To Reproduce

N/A

### Build Environment

```markdown
N/A
```

### Version Information

```text
69e9464117eff84e88b9646c0e9c2e95a807fe60
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

Miss y'all ❤️ Hope everything is going well 😃

TaylorBeebe, 2024-07-27:
@apop5 @makubacki @os-d

os-d, 2024-07-27:
@TaylorBeebe, ha, thanks for catching this. I will fix it.

TaylorBeebe, 2024-07-27:
@os-d

No problem 😄

If you want this test to pass, I think the first step is to look into removing the following lines:
1. https://github.com/microsoft/mu_basecore/blob/856d8a412308b852e53254c10edd0f098933aa8f/MdeModulePkg/Core/Dxe/Misc/MemoryProtectionSupport.c#L1186
2. https://github.com/microsoft/mu_basecore/blob/856d8a412308b852e53254c10edd0f098933aa8f/MdeModulePkg/Core/Dxe/Misc/MemoryProtectionSupport.c#L1221
3. https://github.com/microsoft/mu_basecore/blob/856d8a412308b852e53254c10edd0f098933aa8f/MdeModulePkg/Core/Dxe/Misc/MemoryProtectionSupport.c#L1257

If I remember correctly, one of the issues with removing the above lines is detailed in our Teams chat toward when you joined the team. The project led by Aaron hits a different issue if the above lines are removed which I think he remembers.

I'm pretty limited in what support I can provide, but feel free to reach out if questions arise.

PR #532 (2024-08-28): DxePagingAuditTestApp: Remove MemoryOutsideEfiMemoryMapIsInaccessible Test
## Description

MemoryOutsideEfiMemoryMapIsInaccessible was attempting to test that memory outside the EFI_MEMORY_MAP was marked EFI_MEMORY_RP or unmapped, however this is not a valid test as we expect there to be ranges outside of the EFI_MEMORY_MAP, such as GCD non-existent memory and non-runtime MMIO ranges. This patch removes the test. Closes #531.

- [ ] Impacts functionality?
  - **Functionality** - Does the change ultimately impact how firmware functions?
  - Examples: Add a new library, publish a new PPI, update [... cut]
closed 2024-08-05 by os-d