microsoft/mu_plus #570: [Feature]: Add gBeforeEndOfDxeEventGuid, gAfterEndOfDxeEventGuid and gDispatchDeferredImagesEventGuid and signal it in PlatformBootManagerLib

MarcChen46, 2024-09-03:
### Feature Overview

Due to there are some requirement that we need to invoke some specific functions in some flow that has sequence requirement, we would like to add `gBeforeEndOfDxeEventGroupGuid`, `gAfterEndOfDxeEventGroupGuid` and `gDispatchDeferredImagesEventGroupGuid`.

The `gBeforeEndOfDxeEventGroupGuid` and `gAfterEndOfDxeEventGroupGuid` will be signaled in [ExitPmAuth()](https://github.com/microsoft/mu_plus/blob/release/202405/MsCorePkg/Library/PlatformBootManagerLib/BdsPlatform.c#L33C1-L33C12) that before and after `gEfiEndOfDxeEventGroupGuid`.

The `gDispatchDeferredImagesEventGroupGuid` will be siognaled in [PlatformBootManagerBeforeConsole()](https://github.com/microsoft/mu_plus/blob/release/202405/MsCorePkg/Library/PlatformBootManagerLib/BdsPlatform.c#L175C1-L175C34) after `EfiBootManagerDispatchDeferredImages`

### Solution Overview

- Add gBeforeEndOfDxeEventGuid, gAfterEndOfDxeEventGuid and gDispatchDeferredImagesEventGuid to [MsCorePkg.dec](https://github.com/microsoft/mu_plus/blob/release/202405/MsCorePkg/MsCorePkg.dec)

- The `gBeforeEndOfDxeEventGroupGuid` and `gAfterEndOfDxeEventGroupGuid` will be signaled in [ExitPmAuth()](https://github.com/microsoft/mu_plus/blob/release/202405/MsCorePkg/Library/PlatformBootManagerLib/BdsPlatform.c#L33C1-L33C12) that before and after `gEfiEndOfDxeEventGroupGuid`.

- The `gDispatchDeferredImagesEventGroupGuid` will be siognaled in [PlatformBootManagerBeforeConsole()](https://github.com/microsoft/mu_plus/blob/release/202405/MsCorePkg/Library/PlatformBootManagerLib/BdsPlatform.c#L175C1-L175C34) after `EfiBootManagerDispatchDeferredImages`

### Urgency

Medium

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

MarcChen46, 2024-09-03:
@apop5, please take a look if this proposal is ok to you, then I will find someone to create PR for this proposal, thanks.

apop5, 2024-09-04:
Would it be possible to take this to EDK2 first? Bds is already very fragmented, with each silicon vendor providing their own library instances that implement this additional checks. If EDK2 can align on a common implementation, it would greatly simplify the maintenance, and it could allow silicon vendors to be able to use stock edk2.

MarcChen46, 2024-09-04:
after offline discussion, agree to work with silicon vendor to upstream to edk2, close the issue.