microsoft/mu_plus #636: [Bug]: AddOn VGA display is not working due to GOP from the addon VGA is loaded after console out connection

MarcChen46, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Current Behavior

in current implementation of [PlatformBootManagerLib](https://github.com/microsoft/mu_plus/blob/dev/202405/MsCorePkg/Library/PlatformBootManagerLib/BdsPlatform.c#L269C3-L269C40), the GOP option ROM from addon VGA card is loaded after console out connection, and the addon VGA display is not working due to it, if we adjust the `EfiBootManagerDispatchDeferredImages` prior to `DeviceBootManagerBeforeConsole`, then the issue can be resolved.

In edk2 and edk2-platform repos, there are also some examples of doing console connection after option ROM as below.

[OvmfPkg](https://github.com/tianocore/edk2/blob/bc664d1830c9446cb1d33b10a41e6b3d207997f1/OvmfPkg/Library/PlatformBootManagerLib/BdsPlatform.c#L549-L554)
[UefiPayloadPkg](https://github.com/tianocore/edk2/blob/master/UefiPayloadPkg/Library/PlatformBootManagerLib/PlatformBootManager.c#L277)
[AMD ChachaniBoardPkg](https://github.com/tianocore/edk2-platforms/blob/241cf45003f63b4331f7cc18c8d0d80b7d594367/Platform/AMD/VanGoghBoard/ChachaniBoardPkg/Library/Capsule/PlatformBootManagerLib/PlatformBootManager.c#L509) 

### Expected Behavior

The console out connection should also be performed after GOP option ROM get loaded to make the display from addon VGA work

### Steps To Reproduce

1. System that only has addon VGA card and plug monitor with the addon VGA card.
2. Power on the system
3. Boot to EFI Shell
4. No display from addon VGA

### Build Environment

```markdown
- OS(s): Windows11
- Tool Chain(s): VS2022
- Targets Impacted: all systems that use addon VGA card as primary video controller.
```

### Version Information

```text
v2024050001.0.4
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

MarcChen46, 2025-02-19:
@apop5 @kuqin12 @makuback, please take a look and provide your comment, thanks.

makubacki, 2025-02-19:
@MarcChen46, as mentioned in another thread, a key requirement is that `ExitPmAuth` (`EndOfDxe` & `SmmReadyToLock`) occurs prior to dispatching deferred images. Are there other changes needed in this library instance for your platform or is this the only issue?

MarcChen46, 2025-02-20:
@makubacki, thanks for reply, move both `ExitPmAuth` and `EfiBootManagerDispatchDeferredImages` prior to console out connection is the only required change for resolve the addon VGA display issue.

makubacki, 2025-02-24:
Thanks @MarcChen46, @apop5 should follow up with you soon after an offline discussion within our team.

closed 2025-04-18 by github-actions[bot]