microsoft/mu_plus #665: [Documentation]: Identify improvements for DMA Protection Audit tests

makubacki, 2025-04-08:
### Request Description

Kernel DMA protection requirements allow the OS and firmware to protect the system against malicious and unintended Direct Memory Access (DMA) attacks for all DMA-capable devices. Individual platform requirements are described in https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-kernel-dma-protection.

This repository has DMA protection audit tests in https://github.com/microsoft/mu_plus/tree/dev/202405/UefiTestingPkg/AuditTests/DMAProtectionAudit.

This issue requests that the DMA requirements be reviewed to determine if new checks can be added to the DMA protection audit (or other tests in UefiTestingPkg) to improve the level of detail to users about their platform's compliance to the requirements.

### Are you going to make the change?

Someone else needs to make the change

### Do you need maintainer feedback?

No maintainer feedback needed

eeshanl, 2025-04-15:
# Current DMA Protection Audit tests coverage

#### AMD IVRS (I/O Virtualization Reporting Structure)
  - CheckBMETeardown at ExitBootServices
  - Checks if the IVRS table is present in the ACPI tables and verifies each IVMD memory ranges are marked reserved.
  - Checks if IOMMU is enabled for each status register

#### Intel DMAR
  - CheckBMETeardown at ExitBootServices
  - Similar check as AMD on DMAR table to see if each RMRR memory ranges are marked reserved.
  - Checks if IOMMU is enabled for each status register

# Gaps
  - IoMmu Protocol function tests
  - While this is a good start to verify DMA protection capabilities, these tests are missing actual unit tests around the IoMmu Protocol functions. This needs to be added.

#### Gaps for Arm
  - CheckBMETeardown at ExitBootServices
  - Check if SMMU translation is disabled and SMMU is in bypass mode at ExitBootServices.
  - Similar test as the IVRS/DMAR tests above, we also need to a test for the IORT for Arm.
  - Check the reserved memory ranges if described in the IORT
  - Check if SMMU is enabled for every stream id described in the IORT

kat-perez, 2025-05-12:
@eeshanl hey Eeshan, is there a repository where the code for this is checked in currently? I want to start working on adding functionality for the "Gaps" section. Thanks!

eeshanl, 2025-05-12:
@kat-perez you can find the audit tests here - https://github.com/microsoft/mu_plus/tree/dev/202502/UefiTestingPkg/AuditTests/DMAProtectionAudit

closed 2025-07-03 by mu-automation[bot]